fenugrec/freediag #1: Debug log

ysangkok, 2015-10-27:
In case it is of any interest, I captured a debug log. The program claims that I have a clone chip. Which device is best with FreeDiag, will I get better results if I buy that? I would like to help the project if I can.

```
Script started on Tue 27 Oct 2015 07:38:11 PM CET
janus@Fugl ~/freediag/scantool
 % ./diag_test /dev/ttyUSB0
diag_l1.c:63:  entered diag_l1_init
diag_os_unix.c:434:  WARNING: Not running as superuser; things may not work correctly
diag_l2.c:169:  entered diag_l2_init
clock_gettime(): using CLOCK_MONOTONIC_RAW
clock_nanosleep(): using CLOCK_MONOTONIC
diag_os_gethrt() resolution <= 0us, avg ~0us
diag_os_getms() resolution <= ~1ms, avg ~1ms
testing diag_os_millisleep(), this will take a moment...
diag_os_chronoms() : initial time 2862209049ms; resolution: ~1ms
diag_l2.c:236:  l2_open DUMB on /dev/ttyUSB0, L1proto=128
diag_l1.c:105:  diag_l1_open DUMB on /dev/ttyUSB0 with l1 proto 128
diag_l0_dumb.c:135:  open subinterface /dev/ttyUSB0 protocol 128
diag_tty_unix.c:150:  Device /dev/ttyUSB0 opened, fd 3
diag_tty_unix.c:170:  open: TIOCGSERIAL failed: Inappropriate ioctl for device
diag_tty_unix.c:663:  2862209053 : DTR/RTS changed
open dl0d = 0x97d1240
diag_l2.c:367:  _startCommunications dl0d=0x97d1240 L2proto 3 flags=0x0 10400bps target=0x21 src=0xF1
diag_l2_iso14230.c:520:  _startcomms flags=0x0 tgt=0x21 src=0xF1
diag_l2.c:645:  diag_l2_ioctl 0x97d13b8 cmd 0x2101
diag_tty_unix.c:526:  setup: fd=3, 10400bps, 8 bits, 1 stop, parity 3
diag_tty_unix.c:349:  Speed set using TCSETS + BOTHER.
Warning : speed off by >= 5% !
diag_tty_unix.c:619:  Speed set to 14400, error~38%
diag_l2.c:645:  diag_l2_ioctl 0x97d13b8 cmd 0x2202
diag_l2.c:645:  diag_l2_ioctl 0x97d13b8 cmd 0x2201
diag_l0_dumb.c:532:  device link 0x97d1240 info 0x97d11f0 initbus type 2
diag_l0_dumb.c:352:  slowinit dl0d=0x97d1240 address 0x21
diag_tty_unix.c:526:  setup: fd=3, 5bps, 8 bits, 1 stop, parity 3
diag_tty_unix.c:349:  Speed set using TCSETS + BOTHER.
Warning : speed off by >= 5% !
diag_tty_unix.c:619:  Speed set to 300, error~5900%
diag_l0_dumb.c:457:  _slowinit: address echo error
diag_l0_dumb.c:556:  L0 initbus failed with -1
diag_l0_dumb.c:557: Unspecified Error.
diag_l2.c:688: Unspecified Error.
diag_l2_iso14230.c:721: Unspecified Error.
diag_l2.c:457:  protocol startcomms returned -1
diag_l2.c:463: Unspecified Error.
diag_l2.c:305:  Entered diag_l2_close for dl0d=0x97d1240;
	closing unused dl2link 0x97d10d8.
diag_l2.c:200:  l2_closelink 0x97d10d8 called
diag_l1.c:134:  entering diag_l1_close: ppdl0d=0x97d10d8
diag_l0_dumb.c:186:  l0 link 0x97d1240 closing
Connection to ECU failed (con. (nil))
do_l2_raw_test error. Exiting.
janus@Fugl ~/freediag/scantool
 % ./scantool              
scantool: Interface set to default: CARSIM on /dev/null
scantool: freediag version 1.06
scantool: Type HELP for a list of commands
scantool: Type SCAN to start ODBII Scan
scantool: Then use MONITOR to monitor real-time data
scantool: **** IMPORTANT : this is beta software ! Use at your own risk.
scantool: **** Remember, "debug all -1" displays all debugging info.
hardware interface: use "set interface NAME [dev]" .
NAME is the interface type and [dev] is
a complete device path such as "/dev/ttyS0" or "\\.\COM11"
Valid NAMEs are: 
MET16 BR1 ELM CARSIM DUMB DUMBT 
scantool: Settings loaded from scantool.ini

scantool> set interface elm /dev/ttyUSB0
interface is now ELM on /dev/ttyUSB0
scantool> debug all -1
[... 2059 lines cut ...]
Location of Oxygen Sensors                     -----          
Requesting Mode 7 (Current cycle emission DTCs)...
diag_l3_saej1979.c:211:  send 1 bytes, l2 flags 0x5
diag_l2.c:549:  diag_l2_send 0x85720f0 msg 0xbff661c0 msglen 1 called
diag_l2_iso14230.c:847:  _send: dl2conn=0x85720f0 msg=0xbff661c0 len=1
diag_l0_elm.c:940:  getflags link 0x85615f8 proto 2 flags 0x1C6AB
diag_l1.c:179:  _send: len=1 P4=5 l0flags=0x1C6AB; 0x07 
diag_l0_elm.c:777:  ELM: sending 1 bytes
)iag_l0_elm.c:789:  ELM: (sending string 07
diag_l3_saej1979.c:444:  recv state 4 tout 300
diag_l2.c:614:  diag_l2_recv 0x85720f0 timeout 300 called
diag_l2_iso14230.c:213:  _int_recv dl2conn=0x85720f0 offset=0x0, tout=300
diag_l2_iso14230.c:261:  before recv, state=1 timeout=300, rxoffset 0
diag_l1.c:277:  _recv request len=1024, timeout=300;diag_l0_elm.c:833:  Expecting 3*1024 bytes from ELM, 300 ms timeout(+400)...got 11 bytes, 0x87 0xF1 0x12 0x47 0x00 0x00 0x00 0x00 0x00 0x00 0xD1 
diag_l2_iso14230.c:278:  after recv, rv=11 rxoffset=0
diag_l2_iso14230.c:261:  before recv, state=2 timeout=23, rxoffset 11
diag_l2_iso14230.c:278:  after recv, rv=-8 rxoffset=11
diag_l2_iso14230.c:316:  Copying 11 bytes to data: 0x87 0xF1 0x12 0x47 0x00 0x00 0x00 0x00 0x00 0x00 0xD1 
diag_l2_iso14230.c:261:  before recv, state=3 timeout=150, rxoffset 0
diag_l1.c:277:  _recv request len=1024, timeout=150;diag_l0_elm.c:833:  Expecting 3*1024 bytes from ELM, 150 ms timeout(+400)...^C
janus@Fugl ~/freediag/scantool
 % 

Script done on Tue 27 Oct 2015 07:41:57 PM CET
```

fenugrec, 2015-10-28:
Hi!
Thanks for the report.

> The program claims that I have a clone chip.

Indeed. That detection is based on the version string that your IC returns; I use this information to skip certain commands that are not supported by clones in general (to cut down costs, they do not implement the full ELM327 command set...)

> Which device is best with FreeDiag, will I get better results if I buy that?

That depends what you need. In the log you pasted, it seemed to work OK - the interesting lines are flooded by the very verbose output, but I see at least
````
Requesting Mode 0x01 PID 0x01 (Current DTCs)...
MIL light ON, 1 stored DTC 
Requesting Mode 0x03 (Emission DTCs)...
P1407 
```

1- Is there anything you tried that did not work, or behaved different than expected ? You filed this as an "issue", but I saw no particular problems ?
2- Also, I'd be interested to know what machine + OS + car model + year you were using.

As for recommendations, for the moment I have nothing definite. ELM* chips work OK, but are slow. This is more obvious when trying communications-intensive operations, like monitoring PIDs / live data, etc.
Some improvement is possible on my part, but I can't take many shortcuts because of the way ELMs are programmed. Dumb interfaces *can* be quite a bit faster, but may be slightly less reliable depending factors related to timing (very hard to control). FYI, I use a dumb interface quite successfully with ISO14230 compliant ECUs, but sometimes the init will not work on the 1st attempt.

The best will be J2534-compliant interfaces, but support for those is only at the "planning" stage currently. They would provide the speed of dumb interfaces with the reliability of smart interfaces.

>  I would like to help the project if I can
3- Nice. What hardware do you have access to (computers, cars) ? Just running a few tests now and then, on a different setup from mine, would be helpful to find regressions.
4- Would you be willing to be a packager for your linux distro ? I maintain those for Arch linux but some help on other distros would be very nice ( I run only Arch at the moment).
Thanks!

ysangkok, 2015-10-28:
1 - Everything worked as expected. There were no problems, it is just that there is no way to send private messages on Github.
2 - I tested on Windows 10 and Ubuntu 14.04 LTS. On Ubuntu, I compiled from the current master using cmake. On Windows, I used the newest distributed binaries, v1.06 if I recall correctly.
3 - My car is a Mercedes-Benz W203 (branded as C220 CDI) from 2003 with a 104 KW diesel engine. I would love to run tests for you. My adapter was ordered as "Hochleistung ELM 327 V1.5 OBDII OBD 2 CAN-BUS USB Auto Car Diagnostic Interface Scanner" from amazon.de, but it is not available anymore. It has a blue and orange, it is my impression that many clones look like this.
4 - I am not really interested in packaging, I think most people clever enough to use FreeDiag also have no problems compiling from source using cmake (which is pretty portable).

fenugrec, 2015-10-28:
> There were no problems, it is just that there is no way to send private messages on Github.

Good point, I should add an email contact in the readme. Freediag is also hosted on sourceforge which has a mailing list and forums, but like many other devs I also tend to prefer github for most things.

> I think most people clever enough to use FreeDiag also have no problems compiling from source

Hehe. True, but my cmake setup is not very good at installing freediag. So for example on Arch I have a PKGBUILD script that does the compilation (easy) and some manual file copying because I haven't figured out the best way to do that in cmake. Otherwise "make install" dumps files in stupid places on the filesystem.
So, that's one other place I could use some help : improving cmake install p-)

closed 2015-11-02 by fenugrec