fenugrec/freediag #2: Linking issue whilst building 'scangui'

lag-linaro, 2016-02-22:
Linking CXX executable scangui
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libfltk.a(Fl_x.o): undefined reference to symbol 'dlsym@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libdl.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [scangui/scangui] Error 1
make[1]: *** [scangui/CMakeFiles/scangui.dir/all] Error 2
make: *** [all] Error 2

fenugrec, 2016-02-22:
Interesting... I haven't compiled scangui in a long time. 
Have you tried the obvious, "update fltk to latest" ?

Can you add the following info:

- distro
- fltk version
- gcc version

The issue probably has to do with this :
https://fedoraproject.org/wiki/UnderstandingDSOLinkChange

lag-linaro, 2016-02-22:
Ubuntu 14.04 LTS
FLTK is the latest from the repos [v1.3]
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.1) 4.8.4

Perhaps you just need to explicitly add the library in the Makefiles?

fenugrec, 2016-02-27:
Hi again,
you're not going to like these answers, I apologize in advance !
- I cannot reproduce the issue with gcc-5.3.0 and fltk-1.3.3 !
- I just compiled scangui, and... it's barely a proof-of-concept GUI. It uses the super old style "/dev/obdIIx" symlinks and doesn't let you specify other serial ports or even interface type. I'm pretty sure you'd be disappointed by its functionality !
- I *am* interested in having a better GUI someday, but there are more urgent issues to address in freediag first ! 

That said, I do have a few suggestions :
- "ldd --version", I'd like to know which version of glibc you have on 14.04
- Is there a way for you to test with a more recent fltk build ? I'm not aware of how ubuntu lets you manage that on its LTS releases, it might require manually installing a different version ?
- Can you try a basic test FLTK program, http://www.fltk.org/doc-1.3/basics.html , and compile according to their instructions ? You'll note that their example does not specify "-ldl" on the gcc command line...

fenugrec, 2016-07-15:
@lag-linaro did you try the suggestions I listed ?

lag-linaro, 2016-07-18:
I used a different application in the end.

closed 2016-07-19 by fenugrec