fenugrec/freediag #9: Unorthodox file location

blshkv, 2016-06-19:
Hi,

I found that the software installs and loads number of files into weird locations.

For example, "doc" folder goes into prefix/doc (must be prefix/usr/share/docs/package-version).

The worst is prefix/bin/ directory. The configuration file (*.ini) should go into /etc folder. DB files freediag_carsim*.db should be kept at /var/lib/* (if i'm not wrong)

would be good if you could fix it in a next release.

Thank you

fenugrec, 2016-06-19:
Hi,
thanks for the report. You're right, install is currently a mess. In my defence, the readme has the warning " For the moment, you should not run 'make install' as it will probably
copy files to inappropriate destinations." : )

where I want to go with this is to generate an RC file in $HOME on anything unix-like, but for windows it was more convenient to have a .ini in the executable directory. I'm not sure how to achieve that automatically with cmake.

I package freediag for Arch ,and for that platform I do most of the install manually, like this:
https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=freediag

I'm not sure the examples should go in /var/lib; or at least maybe not on every distro. I would have rather imagined /usr/share/freediag/examples .

I'm very open to suggestions on how to handle install better, but there are a few things on my list I want to fix before that. Patches / PRs are welcome of course.

blshkv, 2016-06-19:
I've just pushed similar ebuild to Pentoo (https://github.com/pentoo/pentoo-overlay/commit/aa22d62bd8cf1c2ee15e3730acf73750b82946bc), thanks for the hint.

The doc location is very easy to fix, but you need to fix a code for .ini file. it should be installed into /etc directory and the tool should look for it over there by default. 

Ideally, we should be able to drop package() or src_install() functions in both distros.

P.S. Don't really care about windows, sorry ;-)

fenugrec, 2016-11-03:
OK, I'm revisiting this now because I would really like to release 1.08 with a sane(r) "install" system.

- the INIFILE option makes the program look for an .ini in the current directory, more for user convenience (those .ini are actually scripts run at startup) than for system-wide configuration
- RCFILE option doesn't install a sample rc file in the user's $HOME, I don't think it should do so at install time (the code currently creates an empty rcfile at startup when there is no rcfile yet)

As for install directories: based on my reading of the FHS and some packages I have on my Arch machine, I think it should look like this on linux:
- /usr/bin/scantool
- /usr/share/freediag/* (README, AUTHORS, etc)
- /usr/share/freediag/doc/* (the html stuff)
- /usr/share/freediag/examples/* (the .db files and sample .ini file)

I considered changing the last three to "/usr/share/freediag-<version>/" like you suggested, but a lot of packages on my system just have <packagename> without the <version> appended, so I'm inclined to go for the shorter version.

blshkv, 2016-11-03:
I suggest to go ahead with any reasonable paths (after reading FHS) and also create variables so that each distro could adjust it if required.

fenugrec, 2016-11-03:
Should be finally fixed now. "cmake -DCMAKE_INSTALL_PREFIX=/usr" will proceed as I describe in my previous comment. Default CMAKE_INSTALL_PREFIX is /usr/local which may be OK on some distros.
The following variables can be set to absolute paths to customize further the destinations:
- BIN_DESTDIR : scantool binary
- DOC_DESTDIR : misc files (README etc) and doc/
- EXAMPLES_DESTDIR : .db and .ini files

@blshkv I quickly checked your pentoo ebuild for freediag, if I did this correctly you'll be able to eliminate the "sed" lines, maybe add -DCMAKE_INSTALL_PREFIX to the cmake args, and also drop the "hax cleanup" at the end.

fenugrec, 2016-12-16:
Probably fixed in 1.08

closed 2016-12-16 by fenugrec

blshkv, 2016-12-16:
Hi, I have just checked. All is almost good. There is an extra 
```
"README_v${PKGVERSIONMAJOR}_${PKGVERSIONMINOR}.txt
```
left in the CMakeLists.txt. This file does not exist anymore and the installation fails. 
Please remove it.

Cheers.

fenugrec, 2016-12-17:
Hmm, I just realized you're using a .tar.gz auto-generated by github, as opposed to the tarballs I manually package (with that README_v1_08.txt file) and release on sourceforge. Not sure what would be more convenient in your case; the sed line you have is OK, but another solution is to use the sf package instead
https://sourceforge.net/projects/freediag/files/freediag/1.0/freediag-1.08-src.tar.gz/download

Due to bad planning on my part, I released 1.08 before fixing this properly (simply marking the file as optional : )
https://github.com/fenugrec/freediag/commit/e3bb82a8e760bbc14d8dd3e9ecd4ad64802fdaea

Hopefully 1.09 will work as-is !!
It should be out within a few weeks, I have some bugs I really want to fix soon.

Thanks for maintaining the package on that distro btw.

blshkv, 2016-12-17:
The changes looks good. No rush with a next release. The functionality is more important.