fenugrec/freediag #12: Assertion that elm chips are on either 38400 or 9600 baud is wrong

Vaelatern, 2016-08-29:
When I `set speed 115200` I expect the code to actually listen to me and try talking at 115200 baud like my connection demands.

fenugrec, 2016-08-29:
it's a bit more complicated than that, because that "set speed" settings is meant for the communications to the ECU, not to the ELM, that's why the help string says "ECU communications speed".
However, you raise a valid point, I see a user-defined setting has become necessary (when I wrote that code all ELMs I knew of where either 9600 or 38400).

Recently @aaeegg added support for 115200 in the autodetection process, can you try most recent git to make sure everything works fine ?
If so, I'll add an ELM-specific configurable item to allow user-defined speeds as well.

Vaelatern, 2016-08-29:
I'll give it a shot and report back.
Thank you for the comment!

fenugrec, 2016-09-30:
Fixed. Use "set speed x" to use custom speeds.

closed 2016-09-30 by fenugrec