fenugrec/freediag #27: Error Message asks for bug report

jackdneilson, 2017-07-15:
Hi there,

I've been trying to use your software with a cheap ELM327 to usb adapter I've bought off of ebay, and I'm having some trouble getting it to connect to my Ford Fiesta. I've posted some of the logging output below.

`scantool> set show
interface: using ELM
Connect speed: 10400
display: metric units
testerid: using 0xF1
addrtype: functional addressing
destaddr: using 0x33
l1protocol: Layer 1 (H/W) protocol to use ISO9141
l2protocol: Layer 2 protocol to use ISO9141
initmode: Initmode to use with above protocol is FAST
L0 options:
	port=/dev/ttyUSB0
	elmspeed=38400
scantool> scan

Trying SAEJ1850-VPW:
clock_gettime(): using CLOCK_MONOTONIC_RAW
clock_nanosleep(): using CLOCK_MONOTONIC
diag_os_gethrt() resolution <= 0us, avg ~0us
diag_os_getms() resolution <= ~1ms, avg ~1ms
testing diag_os_millisleep(), this will take a moment...
diag_os_chronoms() : initial time 1500139781966ms; resolution: ~1ms
diag_l0_elm.c:410:  Sending ATI to ELM32x at 38400...
Clone ELM found, v1.5. Expect inferior performance
L2 connection OK; tring to add SAE J1979 layer...
diag_l2.c:550: Read/Write timeout.
diag_l3.c:301: Read/Write timeout.
diag_l3.c:214: Read/Write timeout.
diag_l3_saej1979.c:728: Read/Write timeout.
diag_l3_saej1979.c:764:  J1979 Keepalive failed ! Try to disconnect and reconnect.
diag_l3_saej1979.c:766: Read/Write timeout.
diag_l3.c:98: Read/Write timeout.
Failed to enable SAEJ1979 mode

Trying SAEJ1850-PWM:
diag_l0_elm.c:410:  Sending ATI to ELM32x at 38400...
diag_l0_elm.c:466:  no valid version string !! Report this !. Got:
>LM37 v1.5
diag_l0_elm.c:469: L0 adapter comms failed.
diag_l2.c:257: L0 adapter comms failed.
scantool.c:714: L0 adapter comms failed.
scantool.c:812: Unspecified Error.
SAEJ1850-PWM Failed!

Trying ISO14230_FAST:
diag_l0_elm.c:410:  Sending ATI to ELM32x at 38400...
Clone ELM found, v1.5. Expect inferior performance
Note : explicit bus init not available on clones. Errors here are ignored.
diag_l0_elm.c:686:  got error while forcing init: ERR
diag_l0_elm.c:687: Unspecified Error.
diag_l0_elm.c:882: Unspecified Error.
diag_l2.c:650: Unspecified Error.
diag_l2_iso14230.c:736: Unspecified Error.
diag_l2.c:438: Unspecified Error.
scantool.c:724:  l2_common_start: l2_StartComm failed
scantool.c:791: Unspecified Error.
ISO14230_FAST Failed!

Trying ISO9141:
diag_l0_elm.c:410:  Sending ATI to ELM32x at 38400...
Clone ELM found, v1.5. Expect inferior performance
Note : explicit bus init not available on clones. Errors here are ignored.
diag_l0_elm.c:686:  got error while forcing init: ERR
diag_l0_elm.c:687: Unspecified Error.
diag_l0_elm.c:882: Unspecified Error.
diag_l2.c:650: Unspecified Error.
diag_l2_iso9141.c:86: Unspecified Error.
diag_l2_iso9141.c:239: Unspecified Error.
diag_l2.c:438: Unspecified Error.
scantool.c:724:  l2_common_start: l2_StartComm failed
scantool.c:761: Unspecified Error.
ISO9141 Failed!

Trying ISO14230_SLOW:
diag_l0_elm.c:410:  Sending ATI to ELM32x at 38400...
Clone ELM found, v1.5. Expect inferior performance
Note : explicit bus init not available on clones. Errors here are ignored.
diag_l0_elm.c:686:  got error while forcing init: ERR
diag_l0_elm.c:687: Unspecified Error.
diag_l0_elm.c:882: Unspecified Error.
diag_l2.c:650: Unspecified Error.
diag_l2_iso14230.c:736: Unspecified Error.
diag_l2.c:438: Unspecified Error.
scantool.c:724:  l2_common_start: l2_StartComm failed
scantool.c:791: Unspecified Error.
ISO14230_SLOW Failed!
scantool.c:1535: Unspecified Error.
Connection to ECU failed
Please check :
	Adapter is connected to PC
	Cable is connected to Vehicle
	Vehicle is switched on
	Vehicle is OBDII compliant
scantool> quit`

fenugrec, 2017-07-15:
Hi,
what year is your Fiesta ? I'm guessing if it's moderately recent, it's probably CAN-only (check the available pins on the OBD connector vs http://freediag.sourceforge.net/OBD_plug_pins_and_protocols.txt ). freediag has no CAN support unfortunately, even if your ELM cable does support it.

About the error message : in your case it's a bit strange, the first character of the version string reply was dropped for some reason, which confused freediag. If you run "scan" a second time, does do the exact same thing ?

jackdneilson, 2017-07-15:
2012, just checked and yep it's CAN. Guess that's something I could contribute during my holiday :)

I've tried to scan another few times and it doesn't give me that error, strange.

Thanks for the fast response!

fenugrec, 2017-07-15:
Ok, I'm closing this until it proves to be repeatable, otherwise it's going to be impossible to track down.
CAN : that would be neat, let me know if you have questions on the codebase !

closed 2017-07-15 by fenugrec