fenugrec/freediag #29: Unable to build, error in CMakeLists.txt

jackdneilson, 2017-09-22:
As the title says, I can't get the program to build. It errors out with two errors;

CMake Error at CMakeLists.txt:117 (install):
  install TARGETS given no RUNTIME DESTINATION for executable target
  "scantool"

CMake Error at CMakeLists.txt:163 (install):
  install FILES given no DESTINATION!

Using gcc version 6.3.0.

fenugrec, 2017-09-22:
Hmm, what version of cmake are you using, what OS, and how are you compiling ?

jackdneilson, 2017-09-22:
CMake ver. 3.7.2-1, OS is Debian Stretch Stable release, compiling by creating a build dir, cd-ing to it then

sudo cmake ../scantool

fenugrec, 2017-09-22:
ah there's you problem; first you shouldn't need to use sudo at all, then (unless I misunderstood) your build dir isn't in the right place, you need to have
./freediag_src/
./freediag_src/scantool (just for reference; you don't access this directly)
./freediag_builddir

then from within freediag_builddir/ you do "cmake ../freediag_src/" etc.

jackdneilson, 2017-09-22:
Ahh, my bad. Thanks for the swift response!

closed 2017-09-22 by jackdneilson