fenugrec/freediag #46: diag_os_unix.c: Posix threads usage, pthread.h

nsajko, 2017-12-29:
The definitions of the diag_os_[a-z]*(mtx|lock) functions are the only thing that depends on <pthread.h> being included in diag_os_unix.c, thus it would be nice to move them out to a separate file; say, diag_os_unix_pthread.c

fenugrec, 2017-12-30:
Is this a real problem though ?
- Win* uses its own native implementations
- OSX has pthread built-in , as I understand
- anything linux/unix should have pthreads built-in

I have never heard of freediag being used on other platforms that fall outside  of those three cases, so I don't see the need to create / split into another category of implementations. Unless the intent is to add C11 atomic-based mutexes, but I see no interest in doing this either until I see serious profiling data that indicates a bottleneck due to mutex overhead P))

nsajko, 2017-12-30:
It's not a "real problem" ;)

I just thought it would be conceptually cleaner not to include a system header in diag_os_unix.c if the relevant functions that use Pthreads can be nicely separated out into their own file.

Nothing important.

fenugrec, 2017-12-30:
Ok, just wanted to make sure I wasn't overlooking an OS/implementation. Closing this since diag_os_unix already #includes  POSIX-only headers (unistd), pthread is not out of place there. Not worth splitting into a separate file IMO - energy is better spent elsewhere !

closed 2017-12-30 by fenugrec