fenugrec/freediag #49: Processes, privilege, and setting realtime scheduling options

nsajko, 2018-01-02:
Scantool needs real time scheduling.

But a program/process that interfaces with external devices, like scantool, should not have the capability to set process scheduling settings; as that is a theoretical security risk and a cautious user could hardly be expected to give scantool root rights without an audit.

This change fixes the issue by introducing another process, meant to be run with root rights and execute scantool safely:       https://github.com/fenugrec/freediag/pull/42

nsajko, 2018-01-02:
The OS did not cause any timing issues for me, I made PR42 simply because scantool principally has the real time requirements, even if they are unlikely to cause a problem on lightly loaded system. But hypothetically one might want to communicate with the car on a more loaded system; perhaps some imagined car computer would simultaneously run scantool, navigation, GPS, render maps, etc.; and in that scenario SCHED_FIFO seems much more important.

nsajko, 2018-01-02:
> big code change that adds complexity that is perhaps not necessary.

Funnily, I thought of it as reducing scantool complexity and adding a simple separate program ;)

fenugrec, 2018-01-02:
>  But hypothetically one might want to communicate with the car on a more loaded system; perhaps some imagined car computer

Ok, yes. Not an interesting use case to me, but valid nevertheless.

> Funnily, I thought of it as reducing scantool complexity and adding a simple separate program

Let me elaborate : current behavior is essentially just the call to sched_setscheduler(getpid(), SCHED_FIFO, &p) . 
I don't believe there is any security risk in doing this, unless scantool is run as superuser - but that should never happen. In fact I'm considering preventing it entirely by exiting if getuid()==0.

The code added in PR 42 is very much more complex. Granted, it is technically a separate program but merging this makes it nominally part of freediag - then thought needs to be given to testing and maintenance etc.

I might be missing something here, but if the intent is just to increase the scantool process priority,
 can't the user (in the hypothetical high-load scenario) just "sudo renice" it to realtime or whatever ? If scantool is exploited, then the malicious code gains a realtime process, but what can it do except hang the system ? And how would schedSetter help in mitigating this ?

nsajko, 2018-01-02:
> complex

Maybe the "wait on FIFO to switch priority" part is not useful enough and should be removed; but I thought the code was quite simple anyways. It just sets priority, in a more POSIX-correct way than diag_os_sched; and forks-drops privileges-execs. But that stuff should probably be discussed on the PR page...

> can't the user

Well, that is kind of what I was going for, but to make it simpler in implementation and less bothersome for the user. The difference in implementation is that schedSetter just executes scantool instead of waiting for a process named scantool to appear, which would be a bit more complex, and it would be required to invoke ps from schedSetter to be POSIX-compliant.

fenugrec, 2018-01-02:
Taking a step back, I think that 

> Scantool needs real time scheduling.

is not exactly the issue then, but rather "freediag doesn't provide a safe/easy way of running scantool in  real-time priority".  Then my main objection would be that this is a corner case, and the user should handle this if/as they require (exec + renice as they need); making this outside the scope of freediag. But you've already written schedSetter, so I guess it could be included as an 'extra' utility.

However, before changing diag_os_unix, could you clarify

> It just sets priority, in a more POSIX-correct way than diag_os_sched; 

What is wrong with the current implementation ? I'm not very well-versed in POSIX but I thought my interpretation was correct.

nsajko, 2018-01-03:
> What is wrong with the current implementation ?

It is just that diag_os_sched hardcodes 1 as sched_priority. I made a "review" at the PR page for comparison:          https://github.com/fenugrec/freediag/pull/42#pullrequestreview-86386001

fenugrec, 2018-01-08:
Interesting discussions, thanks. Fixed in #42 merge, thanks. I'll note here too that on an artificially loaded system here, freediag still runs fine at unpriviledged levels, so I'm still waiting to see a real use case where this is a problem.

closed 2018-01-08 by fenugrec