fenugrec/freediag #53: Freediags Linux binary shouldn't be called scantool

plater, 2019-04-14:
I'm busy packaging freediag for openSUSE (and myself) and I've patched the build to produce /usr/bin/freediag instead of scantool due to the fact that scantool already exists and is available in Ubuntu and debian.
[freediag-rename-scantool.patch.txt](https://github.com/fenugrec/freediag/files/3077422/freediag-rename-scantool.patch.txt)
Besides Linux users are used to names unrelated to functions.
https://build.opensuse.org/package/show/home:plater/scantool

fenugrec, 2019-04-14:
Thanks, I will look at this. It's not the first time packagers have name collisions wiht scantool.net VS freediag.
To be complete however the docs should be patched too, at least the direct references to the CLI binary.

plater, 2019-04-16:
> Thanks, I will look at this. It's not the first time packagers have name collisions wiht scantool.net VS freediag.
> To be complete however the docs should be patched too, at least the direct references to the CLI binary.

I realize that the patch is "WIP", I also plan on working on the can implementation.

fenugrec, 2019-05-21:
I tried applying your patch against c33eab1 :
```
$ git apply freediag-rename-scantool.patch.txt
error: patch failed: CMakeLists.txt:106
error: CMakeLists.txt: patch does not apply
```
what's your patch command to make this work ?

plater, 2019-05-21:
Sorry, it was made with -p0, this one should work after renaming. -p0 is a bad habit I have, git uses -p1
[freediag-rename-scantool.patch.txt](https://github.com/fenugrec/freediag/files/3201381/freediag-rename-scantool.patch.txt)

fenugrec, 2019-05-21:
Thanks, that worked. Applied in 896e75b with doc fixups.

closed 2019-05-21 by fenugrec