fenugrec/freediag #54: I am facing big troubles when attempt to compile with Visual Studio

Fojtik, 2019-10-13:
I am facing big troubles when attempt to compile with Visual Studio.
I could attempt to fix it.

1>------ Skipped Build: Project: RUN_TESTS, Configuration: Debug Win32 ------
1>Project not selected to build for this solution configuration 
2>------ Build started: Project: freediag, Configuration: Debug Win32 ------
2>Linking...
2>scantool.obj : error LNK2019: unresolved external symbol _LL_CONCAT referenced in function _j1979_data_rcv
2>diag.lib(diag_l2.obj) : error LNK2001: unresolved external symbol _LL_CONCAT
2>diag.lib(diag_l3_saej1979.obj) : error LNK2001: unresolved external symbol _LL_CONCAT
2>diag.lib(diag_cfg.obj) : error LNK2001: unresolved external symbol _snprintf
2>diag.lib(diag_l3_iso14230.obj) : error LNK2001: unresolved external symbol _snprintf
2>diag.lib(diag_l3_saej1979.obj) : error LNK2019: unresolved external symbol _snprintf referenced in function _diag_l3_j1979_decode
2>diag.lib(diag_l0_elm.obj) : error LNK2001: unresolved external symbol _snprintf
2>scantool_obd.obj : error LNK2001: unresolved external symbol _snprintf
2>freediagcli.lib(scantool_cli.obj) : error LNK2001: unresolved external symbol _snprintf
2>diag.lib(diag_os_win.obj) : error LNK2001: unresolved external symbol _snprintf
2>diag.lib(diag_dtc.obj) : error LNK2001: unresolved external symbol _snprintf
2>R:\freediag-master\bin\scantool\Debug\freediag.exe : fatal error LNK1120: 2 unresolved externals
2>Build log was saved at "file://r:\freediag-master\bin\scantool\freediag.dir\Debug\BuildLog.htm"
2>freediag - 12 error(s), 0 warning(s)
3>------ Skipped Build: Project: INSTALL, Configuration: Debug Win32 ------
3>Project not selected to build for this solution configuration 
4>------ Skipped Build: Project: PACKAGE, Configuration: Debug Win32 ------
4>Project not selected to build for this solution configuration 
========== Build: 0 succeeded, 1 failed, 5 up-to-date, 3 skipped ==========

fenugrec, 2019-10-13:
I've never used VS myself. Pastebin your "BuildLog.htm" somewhere. Also,
- how did you generate the VS project files
- what version of VS, what value does it define _MSC_VER as

Part of your problems (LL_CONCAT) are related to the #ifdefs in "utlist.h". Hints : check if your compiler defines  LDECLTYPE

_snprintf : you'll have to look at the MSDN docs, and scantool/diag.h:80

Fojtik, 2019-10-13:
#how did you generate the VS project files
It is a standard feature of cmake.

I am using Visual Studio 2008. 

If I fix this I can sens you some patches.

Fojtik, 2019-10-13:
It seems to be a deffect in a code:
LL_CONCAT is defined on line 187 
and UNDEFINED! in line 282.

 R:\freediag-master\scantool\diag_l2.c(159):	LL_CONCAT(d_l2_conn->diag_msg, msg);
  R:\freediag-master\scantool\diag_l3_saej1979.c(383):			LL_CONCAT(d_l3_conn->msg, msg);
  R:\freediag-master\scantool\utlist.h(26): * NOTE !! LL_APPEND() is virtually identical to LL_CONCAT, except
  R:\freediag-master\scantool\utlist.h(186):#define LL_CONCAT(head1,head2)                                                                 \
  R:\freediag-master\scantool\utlist.h(187):    LL_CONCAT2(head1,head2,next)
  R:\freediag-master\scantool\utlist.h(189):#define LL_CONCAT2(head1,head2,next)                                                           \
  R:\freediag-master\scantool\utlist.h(282):#undef LL_CONCAT /* no LL_CONCAT_VS2008 */
  R:\freediag-master\scantool\scantool.c(252):		LL_CONCAT(ep->rxmsg, rmsg);
  Matching lines: 8    Matching files: 4    Total files searched: 85

Fojtik, 2019-10-13:
_snprintf : you'll have to look at the MSDN docs, and scantool/diag.h:80
The problem is clearly described here:
https://stackoverflow.com/questions/21369384/c-error-lnk2019-unresolved-external-symbol-snprintf-referenced-in-function

and the solution is:
#ifdef _MSC_VER	
	#define snprintf _snprintf	//danger : _snprintf doesn't guarantee zero-termination !?
#endif

I have modified utlist.h - I would like to send you fixed header file.

Fojtik, 2019-10-13:
I cannot upload .h file. File mustbe zipped, because AI is stupid:
[utlist.zip](https://github.com/fenugrec/freediag/files/3722166/utlist.zip)

fenugrec, 2019-10-13:
I will look at utlist.h later, a cleaner fix will be for me to update to the latest utlist.h from upstream ( https://github.com/troydhanson/uthash/blob/master/src/utlist.h ). Try replacing that file, see if it fixes the problem.

_MSC_VER : you're probably right. MSVC is something defined at the CMake level but I don't think it's automatically #defined ... You're one of the few who have ever tried compiling freediag in MSVC, I have never tried !

PS the "AI" may be stupid, but it's not how your supposed to use github. you usually fork the project, create a branch for your fixes, do the tests/commits *in your branch*, then link to those or do a PR.

Fojtik, 2019-10-13:
If I define MSVC, the project totally fail to compile.

I have even much more fixes to make MSVC compilation working. I can commit them somewhere. the utlist.h needed to write severallines of a new macro thathas been missing for VisualStudio.

I am pretty frustrated from compilation result:
13.10.2019  21:13           194 048 freediag.exe
03.11.2016  23:59           548 703 scantool.exe

New build is 3 times shorter than official installation. I have no idea why. May be that official installation has buildin runtime.

What about to mane a new Windows release & installation?

fenugrec, 2019-10-13:
> If I define MSVC, the project totally fail to compile.

I didn't say you should do that, but it might indicate other problems. If you don't post the errors I can't guess what they are.

Are you saying that modifying  diag.h:80  as
`#ifdef _MSC_VER`
doesn't work ? if not, **paste the errors somewhere**, I can't guess, and I can't try.

> frustrated from compilation result

why ? isn't a smaller binary better ? the official release is compiled with mingw-gcc which pulls in some extra stuff, like for example a C99-compliant snprintf().

Did you try the updated utlist.h ? Fix one thing at a time, otherwise I'm not going to merge anything.

Fojtik, 2019-10-13:
The proposed patch in utililist.h is innocent and will not break further merge.

You asked for errors, here they are:
```
1>------ Skipped Build: Project: RUN_TESTS, Configuration: Release Win32 ------
1>Project not selected to build for this solution configuration 
2>------ Build started: Project: diag, Configuration: Release Win32 ------
3>------ Build started: Project: dyno, Configuration: Release Win32 ------
3>Compiling...
2>Compiling...
3>dyno.c
2>diag_cfg.c
3>r:\freediag-master\scantool\diag.h(85) : error C2628: 'SSIZE_T' followed by 'int' is illegal (did you forget a ';'?)
3>r:\freediag-master\scantool\diag.h(85) : warning C4091: 'typedef ' : ignored on left of 'LONG_PTR' when no variable is declared
3>Warning: MSVC _sprintf() may be dangerous !     Please ask your compiler vendor to supply a C99-compliant snprintf()...
2>r:\freediag-master\scantool\diag.h(85) : error C2628: 'SSIZE_T' followed by 'int' is illegal (did you forget a ';'?)
2>r:\freediag-master\scantool\diag.h(85) : warning C4091: 'typedef ' : ignored on left of 'LONG_PTR' when no variable is declared
2>Warning: MSVC _sprintf() may be dangerous !     Please ask your compiler vendor to supply a C99-compliant snprintf()...
2>Warning: MSVC may not work with the CURFILE macro. See diag.h
3>Warning: MSVC may not work with the CURFILE macro. See diag.h
2>r:\freediag-master\scantool\diag.h(96) : warning C4005: 'CURFILE' : macro redefinition
3>r:\freediag-master\scantool\diag.h(96) : warning C4005: 'CURFILE' : macro redefinition
2>        command-line arguments :  see previous definition of 'CURFILE'
3>        command-line arguments :  see previous definition of 'CURFILE'
2>..\..\scantool\diag_cfg.c(64) : warning C4996: 'strcpy': This function or variable may be unsafe. Consider using strcpy_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
2>..\..\scantool\diag_cfg.c(243) : warning C4996: 'strcpy': This function or variable may be unsafe. Consider using strcpy_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
3>Build log was saved at "file://r:\freediag-master\bin\scantool\dyno.dir\Release\BuildLog.htm"
2>..\..\scantool\diag_cfg.c(325) : warning C4996: 'strcpy': This function or variable may be unsafe. Consider using strcpy_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
2>..\..\scantool\diag_cfg.c(329) : warning C4996: 'strcpy': This function or variable may be unsafe. Consider using strcpy_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
2>Build log was saved at "file://r:\freediag-master\bin\scantool\diag.dir\Release\BuildLog.htm"
2>diag - 1 error(s), 6 warning(s)
3>dyno - 1 error(s), 2 warning(s)
4>------ Build started: Project: freediagcli, Configuration: Release Win32 ------
4>Compiling...
4>scantool_cli.c
4>r:\freediag-master\scantool\diag.h(85) : error C2628: 'SSIZE_T' followed by 'int' is illegal (did you forget a ';'?)
4>r:\freediag-master\scantool\diag.h(85) : warning C4091: 'typedef ' : ignored on left of 'LONG_PTR' when no variable is declared
4>Warning: MSVC _sprintf() may be dangerous !     Please ask your compiler vendor to supply a C99-compliant snprintf()...
4>Warning: MSVC may not work with the CURFILE macro. See diag.h
4>r:\freediag-master\scantool\diag.h(96) : warning C4005: 'CURFILE' : macro redefinition
4>        command-line arguments :  see previous definition of 'CURFILE'
4>..\..\scantool\scantool_cli.c(714) : error C2065: 'FL' : undeclared identifier
4>..\..\scantool\scantool_cli.c(821) : warning C4996: 'strcpy': This function or variable may be unsafe. Consider using strcpy_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
4>..\..\scantool\scantool_cli.c(822) : warning C4996: 'strcat': This function or variable may be unsafe. Consider using strcat_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
4>..\..\scantool\scantool_cli.c(826) : error C2065: 'FL' : undeclared identifier
4>Build log was saved at "file://r:\freediag-master\bin\scantool\freediagcli.dir\Release\BuildLog.htm"
4>freediagcli - 3 error(s), 4 warning(s)
5>------ Build started: Project: freediag, Configuration: Release Win32 ------
5>Compiling...
5>scantool_vag.c
5>r:\freediag-master\scantool\diag.h(85) : error C2628: 'SSIZE_T' followed by 'int' is illegal (did you forget a ';'?)
5>r:\freediag-master\scantool\diag.h(85) : warning C4091: 'typedef ' : ignored on left of 'LONG_PTR' when no variable is declared
5>Warning: MSVC _sprintf() may be dangerous !     Please ask your compiler vendor to supply a C99-compliant snprintf()...
5>Warning: MSVC may not work with the CURFILE macro. See diag.h
5>r:\freediag-master\scantool\diag.h(96) : warning C4005: 'CURFILE' : macro redefinition
5>        command-line arguments :  see previous definition of 'CURFILE'
5>Build log was saved at "file://r:\freediag-master\bin\scantool\freediag.dir\Release\BuildLog.htm"
5>freediag - 1 error(s), 2 warning(s)
6>------ Skipped Build: Project: INSTALL, Configuration: Release Win32 ------
6>Project not selected to build for this solution configuration 
7>------ Skipped Build: Project: PACKAGE, Configuration: Release Win32 ------
7>Project not selected to build for this solution configuration 
========== Build: 0 succeeded, 4 failed, 2 up-to-date, 3 skipped ==========
```

I am using this fix instead:
```
[... cut ...]
	//CURFILE will be defined by CMake on a per-file basis !
	#pragma message("Warning: MSVC may not work with the CURFILE macro. See diag.h")
	// apparently some (all ?) MSVC compilers don't support per-file defines, so CURFILE would be the
	// same for all source files.
	// The disadvantage of __FILE__ is that it often (always ?) holds the absolute path of the file,
	// not just the filename. For our debugging messages we only care about the filename, hence CURFILE.
	#define CURFILE __FILE__
#else
	#define FL CURFILE, __LINE__
#endif
```

fenugrec, 2019-10-13:
Thanks. For the moment I'll ignore some of the warnings in there;

- Can you still try the upstream utlist.h I posted, I **need** to know if it works as-is.

- can you try this in diag.h to replace that same block you posted:
```
#ifdef _MSC_VER
	typedef SSIZE_T ssize_t;
	#define snprintf _snprintf	//danger : _snprintf doesn't guarantee zero-termination !?
	#define FL  __FILE__, __LINE
#else
	#define FL CURFILE, __LINE__
#endif
```
That should take care of the "CURFILE redefined" warning as well as the "FL undeclared identifier" error

Fojtik, 2019-10-13:
Yes, the utlist.h seems to compile smoothly.

========== Rebuild All: 6 succeeded, 0 failed, 3 skipped ==========

Fojtik, 2019-10-13:
#define FL  __FILE__, __LINE
   does not work!

6>..\..\scantool\scantool_dyno.c(137) : error C2065: '__LINE' : undeclared identifier
6>..\..\scantool\scantool_dyno.c(292) : error C2065: '__LINE' : undeclared identifier
6>..\..\scantool\scantool_dyno.c(439) : error C2065: '__LINE' : undeclared identifier

use rather:
	#define FL  __FILE__, __LINE__

fenugrec, 2019-10-13:
Oops! I cut that line a bit short. Indeed of course it needs to be __LINE__. I've just pushed some fixes for those two issues (#if _MSC_VER, and new utlist.h).

If you're still worried that the binary is missing parts, try running "ctest -I 3 -j 8" from your build directory, it will run some selftests which should pass.

Fojtik, 2019-10-13:
I have no recipe in CMakeLists.txt  to build ctest.exe.

But the worst think is that program does not communicate with my car.

fenugrec, 2019-10-13:
ctest is part of, and included with, cmake.

I'll close this since the main issue (not compiling) is solved, you can open another one for the failed comms

closed 2019-10-13 by fenugrec

Fojtik, 2019-10-13:
Yes, close it. For other issues it is wise to open a new thread.