fenugrec/freediag #56: Another compilation issues, that I miss to report

Fojtik, 2019-10-14:
inside diag_general.c:
#ifdef _MSC_VER
 #define ASSIGN_ARRAY(ARR_POSITION)
#else
 #define ASSIGN_ARRAY(ARR_POSITION) [ ARR_POSITION ] =
#endif

const char *dbg_prefixes[] = {
	ASSIGN_ARRAY(DIAG_DEBUGPF_NONE)   "GLOB:",
	ASSIGN_ARRAY(DIAG_DEBUGPF_OPEN)   "OPEN:",
	ASSIGN_ARRAY([DIAG_DEBUGPF_CLOSE) "CLOSE:",
	ASSIGN_ARRAY(DIAG_DEBUGPF_READ)   "READ:",
	ASSIGN_ARRAY(DIAG_DEBUGPF_WRITE) "WRITE:",
	ASSIGN_ARRAY(DIAG_DEBUGPF_IOCTL)  "IOCTL:",
	ASSIGN_ARRAY(DIAG_DEBUGPF_PROTO)  "PROTO:",
	ASSIGN_ARRAY(DIAG_DEBUGPF_INIT)   "INIT:",
	ASSIGN_ARRAY(DIAG_DEBUGPF_DATA)   "DATA:",
	ASSIGN_ARRAY(DIAG_DEBUGPF_TIMER)  "TIMER:",
};

inside diag_l2.c
/* struct to manage L2 stuff, used in here only */
static struct {
	diag_mtx connlist_mtx;            // mutex for accessing dl2conn_list
	struct diag_l2_conn *dl2conn_list; // linked-list of current diag_l2_conn-s
	struct diag_l2_link *dl2l_list;    // linked-list of current L2-L0 links
	bool init_done;
} l2internal = {
#ifdef _MSC_VER
        NULL, NULL, NULL
#else
	.dl2conn_list = NULL,
	.dl2l_list = NULL,	// linked-list of current L2-L0 links
	.init_done = NULL
#endif
};

fenugrec, 2019-10-14:
Those are called "designated initializers" , standard since C99 ( https://en.cppreference.com/w/c/language/struct_initialization ). MSVC refuses to support C99 unless that changed in recent versions. You may encounter more problems too.

I'm not sure how many hacks I want to add just to support MSVC and its limitations; can't you use clang or mingw-gcc?

Fojtik, 2019-10-14:
It is up to you, which patches you will use. These hints were only for you to improve your code.

If freediag cannot connect to my car I cannot use it anyway.

fenugrec, 2019-10-14:
Yes, thanks for reporting this, I will keep it open for reference.

fenugrec, 2021-02-07:
I think this is probably fixed with PR #62 and recent versions of MSVC.

closed 2021-02-07 by fenugrec