fenugrec/freediag #60: Useless call for reading Mode 9 CVN

Elmue, 2020-09-22:
scantool_test.c  
Line 159:

```
if (merged_mode9_info[6]) {
		**get_vit_info**(d_conn, 6, infostring, MODE9_INFO_MAXLEN);
		unsigned cvn_len = **get_vit_info**(d_conn, 6, infostring, MODE9_INFO_MAXLEN);
		if (cvn_len) {
			printf("CVN: ");
			diag_data_dump(stdout, infostring, cvn_len);
			printf("\n");
		}
	} else {
		printf("ECU doesn't support CVN request\n");
	}
```
Can anybody explain me why get_vit_info() is called twice with the same parameters ?
If this is a bug the first call should be removed.
If this  is by purpose a comment should be added which explains the sense of the second call.

fenugrec, 2020-09-22:
I don't think that's on purpose. Can you remove the first call and test

Elmue, 2020-09-24:
I downloaded the Release version.
I cannot compile the project on Visual Studio so I cannot test it.
But I see that the command is sent twice in a data logger.
And the code shows that the result of the first call is not used in any place.
It seems to be a command that someone has put fort testing and forgot to remove it.

fenugrec, 2021-01-31:
Fixed in https://github.com/fenugrec/freediag/commit/020079de45446fbef417bef42b237c45248deece

closed 2021-01-31 by fenugrec