fenugrec/freediag #65: CI / continuous testing would be nice

fenugrec, 2021-02-11:
from PR #62  , the ability to test compilation on MSVC + borland + mingw automatically would be nice.

cfehse, 2021-02-12:
> test compilation on MSVC + borland + mingw

- Borland could be a bit of a problem probably because it may not be installed on the test runner.
- Linux build should be tested as well I think.

> running with valgrind for bonus points
> cppcheck / other static checker for more bonus

Yes that would be done in a secound step.

cfehse, 2021-02-25:
@fenugrec 

Which kind of approach would you prefer for the github action workflow:

A more "native" and "splitted" approach as for example here:
[https://github.com/danmar/cppcheck/tree/main/.github/workflows](url)

Or a cmake based "all in one" definition as for example here: [https://github.com/cristianadam/HelloWorld/tree/master/.github/workflows](url)

fenugrec, 2021-02-27:
> A more "native" and "splitted" approach as for example here:

If it's a comparable amount of work and functionality, I think I prefer the "split" approach - a bit easier to digest and work on, perhaps ?
If there are reasons to prefer the second, I could be convinced.

cfehse, 2021-02-27:
I just asked for personal preference. I would tend to "split" approach as well. 

So decision made: 1 workflow definition per architecture.