fenugrec/freediag #97: Two-byte Volvo DTCs (MY99+)

aaeegg, 2024-03-29:
According to Aleksi Venäläinen, for the SRS and Denso ECUs on model year '99 and '00, the second byte of readDiagnosticTroubleCodes and clearDiagnosticInformation requests should be 0x11, and the ECU will respond with two bytes per DTC rather than a single byte. The two-byte DTC hex values are used directly as the suffix of the human-readable DTC number. Presumably this also applies to model year '01 and later vehicles, but those are CANbus and we currently can't talk to them at all.

Example request/response provided by Aleksi:

> AE11 [plus checksum]
> 87 13 7A EE 11 51 0D 14 00 85

Decodes as:

> ECM-510D Vehicle Speed signal. Faulty Signal
> ECM-1400 Manifold absolute pressure (MAP) Signal High / Low

I don't have a MY99/00 ECU. @brendandburns do you have a model year '99 or '00 car you can test with?

brendandburns, 2024-04-01:
Sadly mine is a 1998 V70 so I can't test either.

aaeegg, 2024-04-02:
This will have to wait until someone with a suitable car shows up or someone gets a bench ECU with two-byte DTCs.

aaeegg, 2024-04-07:
ECUs using two-byte DTCs will respond to AE 01 (request for single byte DTC) with an error code.

SRS ECUs with P/N 9472727 and 8622314, made by TEMIC, might be suitable for testing.